dji-sdk/Onboard-SDK #866: When we executed a mission on the M300, the drone went into a hovering state after taking off.

btsatoukzn, 2022-01-26:
hi,

we use OSDK4.1.0.
After starting MatriceSeries_ConfigTool, when djiosdk-waypoint-v2-sample was executed continuously, it often happened that the hovering state occurred after takeoff.
In some cases, the event occurred in the second mission execution, and in other cases, it occurred in the third mission.

We are worried that this event will occur on flights using real aircraft.
Does this happen when using the simulator?

thank you.

dji-dev, 2022-01-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59738'>Zendesk ticket #59738</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

May I ask that you are using MatriceSeries_ConfigTool to start the simulator? For M300, you can start the simulator with DJI Assistant2, don't start MatriceSeries_ConfigTool.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2022-01-28:
thank you for your reply.

With your advice, We used DJI Assistant2(Enterprise Series)(V2.1.2) and the results were the same.
After executing a mission ending in RTL, when the mission was executed again, it became hovering after takeoff.

After starting the drone, the following operations worked.

1. upload mission
2. start mission
3. stop mission
4. upload mission
5. start mission
 (After that, repeat steps 3-5. Repeated 10 times. )

We think that the event will occur when the second mission is executed from the ground.

dji-dev, 2022-01-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59738'>Zendesk ticket #59738</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Did you disconnect the remote control? If the remote controller has been disconnected, when executing the task for the second time, please make sure that the OSDK has gained control authority again.
  vehicle->control->obtainCtrlAuthority(functionTimeout);

And enable  SDK failsafe action on DJI Assistant2.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2022-01-31:
thank you for your reply.

We think that the remote controller is not disconnected.

With your advice, We added obtain control before start mission. but the result was same.

As a test, we changed the failsafe action to rtl and ran the mission. but the result was same.
After executing a mission ending in RTL, when the mission was executed again, it became hovering after takeoff. didn't RTL.
[image]

dji-dev, 2022-01-31:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59738'>Zendesk ticket #59738</a>:<br><br>please check the email of ticket:59740
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2022-02-02:
thank you for your reply.

We have confirmed that the following error message is displayed on the remote controller when the hover state is reached after takeoff in the second and subsequent mission executions. And this error message disappears immediately.

> Obstacle detection. Pause flight path mission.

On the remote controller, we have obstacle detection turned off.
Why is the obstacle detection error message displayed?
How can we prevent this error message from being displayed?

[image]

dji-dev, 2022-02-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59738'>Zendesk ticket #59738</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The remote control error shows that obstacle avoidance has been triggered, as the waypoint mission detects obstacles, especially when flying upwards. Your phenomenon appears to be an obstacle or blockage to the upper vision sensor. You can check the upper sensor or wipe the sensor to avoid false triggering of obstacle avoidance due to dirt.
In addition, if there are other drones, you can also replace the drone to test whether it is normal. Check whether it is related to the hardware of the drone.

Please follow up on this issue in mail ticket: 59740

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2022-02-09:
Thank you for your reply.

Yesterday, we conducted an M300 mission flight outdoors.
First, we wiped the top sensor with a soft cloth.
Then we ran the mission.
We were able to execute the mission four times in a row.

thank you.

closed 2022-02-09 by btsatoukzn