dji-sdk/Onboard-SDK #870: DisplayMode = 52.

IvanOrfanidi, 2022-02-16:
Hi, friend!
We had a problem with the M300 drone.
We couldn't cancel the drone's homecoming. We got an error that the drone was in DisplayMode = 52.

Here is the text:
**During landing, the drone DJI Matrice M300 with the manifold 2G failed to interrupt RTH. At the same time, the drone was in Display mode = 52. RTH at this moment was not interrupted by a command from the OSDK.
Please advise what is display mode=52**

Tell me what is displayMode = 52? How to cancel the landing if we are in this mode. We could not find a description of this mode:
_https://developer.dji.com/onboard-api-reference/namespaceDJI_1_1OSDK_1_1VehicleStatus_1_1DisplayMode.html_

Thank you for understanding, friend.

dji-dev, 2022-02-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/60349'>Zendesk ticket #60349</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

 **display mode=52** means the drone enter three-propeller emergency landing，
This is an emergency landing status. Please check the drone status through the remote control and Pilot APP. If necessary, please provide the drone log and contact support@dji.com to apply for after-sales service of the drone.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/aCCZgGPiX2bkfxq8JG69cAI43' target='zd_attach'>inline2092101867.png</a>
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-02-17:
Thank you dear Friend!

We have already contacted the dealer.

I'm glad you always answer my questions.

closed 2022-02-17 by IvanOrfanidi