dji-sdk/Onboard-SDK #872: Waypoint Mission unrecorded home point

wongkenchiang97, 2022-02-19:
I'm testing DJI demo_mission with below setups:
1. ROS OSDK version: 3.8.1
2. OS: Ubuntu 20.04
3. Flight Controller(DJI M100): N1 

My waypoints are start location->selected goal location->start location. 3 waypoints.
[image]

Terminal output of demo_mission node. Warning message:"Failed sending mission start command".
[image]

Terminal output of DJI ROS SDK node. ROS OSDK status:"MISSION_UNRECORDED_HOME".
[image]

I've tried to add home point on DJI ROS SDK node's code referring to #526 issue, but SDK package don't have 
"<dji_flight_controller.hpp>" header file.

Anyone have solved this issue? @rohitsantdji

dji-dev, 2022-02-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/60453'>Zendesk ticket #60453</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The info you provide shows that the latitude and longitude of your waypoint is an angle value. please upload with radians.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/hv5fa8X0h2vouyXuEsb4T64Iu' target='zd_attach'>inline-970578985.png</a>
<br><p><em><small>°°°</small></em></p>

wongkenchiang97, 2022-02-21:
> Agent comment from kyle.cai in [Zendesk ticket #60453](https://djisdksupport.zendesk.com/agent/tickets/60453):Dear developer ,
> 
> Thank you for contacting DJI.
> 
> The info you provide shows that the latitude and longitude of your waypoint is an angle value. please upload with radians. [image]​
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards, DJI SDK Support Team[inline-970578985.png](https://djisdksupport.zendesk.com/attachments/token/hv5fa8X0h2vouyXuEsb4T64Iu)
> 
> _°°°_

Thanks for your kind reply and reminder!!