dji-sdk/Onboard-SDK #878: Switch the video mode from the h20 camera to the BGR format.

IvanOrfanidi, 2022-03-28:
Hello!
Can You help me?
I want to switch the video mode from the camera H20.
Now mode is **RGBA**
_converterToAppSink = gst_caps_new_simple("video/x-raw", "format", G_TYPE_STRING, "RGBA", nullptr);_

How can I switch the H20 to BGR?

Thank you for your help!

IvanOrfanidi, 2022-03-28:
void initElements() {
    pipeline = gst_pipeline_new("mypipeline");
    if (pipeline == nullptr) { throw std::runtime_error("failed to create pipeline"); }

    appSrc = gst_element_factory_make("appsrc", "source");
    if (appSrc == nullptr) { throw std::runtime_error("failed to create app source"); }

    h264InputParser = gst_element_factory_make("h264parse", "input h264 parser");
    if (h264InputParser == nullptr) { throw std::runtime_error("failed to create input h264 parser"); }

    decoder = gst_element_factory_make("omxh264dec", "decoder");
    if (decoder == nullptr) { throw std::runtime_error("failed to create decoder"); }

    decodeConverter = gst_element_factory_make("nvvidconv", "decode converter");
    if (decodeConverter == nullptr) { throw std::runtime_error("failed to create decode converter"); }

    loop = g_main_loop_new(nullptr, false);
    if (loop == nullptr) { throw std::runtime_error("failed to create loop"); }

    bus = gst_pipeline_get_bus(GST_PIPELINE(pipeline));
    if (bus == nullptr) { throw std::runtime_error("failed to get bus from pipeline"); }

    appSink = gst_element_factory_make("appsink", "app sink");
    if (appSink == nullptr) { throw std::runtime_error("failed to create app sink"); }
  }

  void initCaps() {
    srcToParser =
      gst_caps_new_simple("video/x-h264", "stream-format", G_TYPE_STRING, "byte-stream", "alignment", G_TYPE_STRING, "nal", nullptr);
    if (srcToParser == nullptr) { throw std::runtime_error("failed to init source to parser caps"); }

    converterToAppSink = gst_caps_new_simple("video/x-raw", "format", G_TYPE_STRING, "RGBA", nullptr);
    if (converterToAppSink == nullptr) { throw std::runtime_error("failed to create converter to sink caps"); }

    nvvidconvToConverter = gst_caps_new_empty_simple("video/x-raw");
    if (nvvidconvToConverter == nullptr) { throw std::runtime_error("failed to create converter to sink caps"); }

    auto feature = gst_caps_features_new("memory:NVMM", nullptr);
    if (feature == nullptr) { throw std::runtime_error("failed to init feature"); }

    gst_caps_set_features(nvvidconvToConverter, 0, feature);
  }

IvanOrfanidi, 2022-03-28:
auto self = static_cast<Decoder*>(data);
    if (self == nullptr) { return GST_FLOW_EOS; }

    GstSample* sample = gst_app_sink_pull_sample(GST_APP_SINK(appSink));
    GstCaps* caps = gst_sample_get_caps(sample);
    GstBuffer* buffer = gst_sample_get_buffer(sample);
    GstStructure* structure = gst_caps_get_structure(caps, 0);
    const int width = g_value_get_int(gst_structure_get_value(structure, "width"));
    const int height = g_value_get_int(gst_structure_get_value(structure, "height"));
    g_value_get_string(gst_structure_get_value(structure, "format"));

    if (!self->capsPrinted) {
      INFO("caps: %s\n", gst_caps_to_string(caps));
      self->capsPrinted = true;
    }

    // Get frame data
    GstMapInfo map;
    if (!gst_buffer_map(buffer, &map, GST_MAP_READ)) { return GST_FLOW_ERROR; }

    cv::Mat img(cv::Size(width, height), CV_8UC4, map.data);

    cv::cvtColor(img, img, CV_RGBA2BGR);

    self->decodedFrameQueue.push(img);

    gst_buffer_unmap(buffer, &map);
    gst_sample_unref(sample);

dji-dev, 2022-03-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/62150'>Zendesk ticket #62150</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The video format OSDK get from the drone is H264, and sample decode the H264 to RGB using ffmpeg, you can change this part of code.
https://github.com/dji-sdk/Onboard-SDK/blob/add352ea12e202b639d4dc84d9bb6beb416f4db0/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.cpp#L237

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-03-29:
Thanks!
I am always glad of your support.

closed 2022-03-29 by IvanOrfanidi