dji-sdk/Onboard-SDK #889: error: field 'currentLogFilePath' has incomplete type 'std::string'

jaegersb, 2022-07-19:
Hello,

I am trying to compile the OSDK on a Raspberry Pi4. The hardware is fresh out of the box, running the latest software versions and I installed all libraries as I went along the installation process, incl. build-essential, etc.

It all works until I run make djiosdk-core. At 78% it pops below claiming an incomplete type std::string for the logfile path. 

Any help is appreciated. Thanks - JJ

[ 78%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/modules/src/filemgr/impl/mmap_file_buffer.cpp.o
In file included from /home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:4:
/home/admin/Onboard-SDK/osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:23:15: error: field 'currentLogFilePath' has incomplete type 'std::string' {aka 'std::__cxx11::basic_string<char>'}
   23 |   std::string currentLogFilePath;
      |               ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/10/iosfwd:39,
                 from /usr/include/c++/10/memory:74,
                 from /home/admin/Onboard-SDK/osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:13,
                 from /home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:4:
/usr/include/c++/10/bits/stringfwd.h:74:11: note: declaration of 'std::string' {aka 'class std::__cxx11::basic_string<char>'}
   74 |     class basic_string;
      |           ^~~~~~~~~~~~
/home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:15:39: error: 'path' has incomplete type
   15 | bool MmapFileBuffer::init(std::string path, uint64_t fileSize) {
      |                           ~~~~~~~~~~~~^~~~
In file included from /usr/include/c++/10/iosfwd:39,
                 from /usr/include/c++/10/memory:74,
                 from /home/admin/Onboard-SDK/osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:13,
                 from /home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:4:
/usr/include/c++/10/bits/stringfwd.h:74:11: note: declaration of 'std::string' {aka 'class std::__cxx11::basic_string<char>'}
   74 |     class basic_string;
      |           ^~~~~~~~~~~~
/home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp: In member function 'bool DJI::OSDK::MmapFileBuffer::init(std::string, uint64_t)':
/home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:19:3: error: 'printf' was not declared in this scope
   19 |   printf("Preparing File : %s\n", this->currentLogFilePath.c_str());
      |   ^~~~~~
/home/admin/Onboard-SDK/osdk-core/modules/src/filemgr/impl/mmap_file_buffer.cpp:7:1: note: 'printf' is defined in header '<cstdio'; did you forget to '#include <cstdio>'?
    6 | #include <string.h>
  +++ |+#include <cstdio>
    7 | 
make[3]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/build.make:524: osdk-core/CMakeFiles/djiosdk-core.dir/modules/src/filemgr/impl/mmap_file_buffer.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:580: osdk-core/CMakeFiles/djiosdk-core.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:587: osdk-core/CMakeFiles/djiosdk-core.dir/rule] Error 2
make: *** [Makefile:183: djiosdk-core] Error 2

dji-dev, 2022-07-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/67906'>Zendesk ticket #67906</a>:<br><br>Use g++ version 5.4 or above and try to compiled with C++11.
<br><p><em><small>°°°</small></em></p>

jaegersb, 2022-07-22:
Hello, I am using g++ version 10.2.1, which is the latest available for Raspian, but still having the same issue.

admin@sophia:~/Onboard-SDK/build $ g++ --version
g++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110

Thank you JJ

jaegersb, 2022-07-26:
Hello,

I have tried with Ubuntu 22.04 and got the same result.
It works however if you follow through on Ubuntu 16.04 with gcc 5.4.0. 

The working combinations I found out are:
g++ (Debian 8.3.0-6) 8.3.0 on Debian GNU/Linux 10 (buster) - 100%
g++ (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609 on Ubuntu 16.04 LTS - requires manual OpenCV upgrade to v3.4, fails at 99% due to No such file or directory (to be investigated)

Thanks for the help JJ

closed 2022-07-26 by jaegersb