dji-sdk/Onboard-SDK #893: M300 firmware 04.01.0000

durka, 2022-08-25:
Does OSDK 4.1.0 support the latest M300 firmware 04.01.0000? Are there any changes I should be aware of?

dji-dev, 2022-08-26:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/69730'>Zendesk ticket #69730</a>:<br><br>M300 firmware 04.01.0000 can be adapted to OSDK 4.x. OSDK 4.x needs to set the baud rate on DJI Assistant2, and the baud rate should same as OSDK settings.
<br><p><em><small>°°°</small></em></p>

Barathwaja, 2022-09-21:
@dji-dev  - If this supports 04.01.0000 then why is the firmware compatibility section in readme shows old versions and the sentences says as latest version?

Please update the README according to the versions.

durka, 2022-12-12:
@dji-dev same question re the new v05.01.01.00 firmware release. There is no mention of OSDK in the release notes and no mention of v05.01.01.00 in the README here.

As the firmware cannot be downgraded now, I am worried about upgrading our fleet without confirmation that it will still be compatible with my OSDK software.

**Are there any changes between 02.04.0102 and 05.01.01.00 that affect OSDK compatibility?**

durka, 2023-02-15:
Empirically, it works.

closed 2023-02-15 by durka