dji-sdk/Onboard-SDK #895: ErrorCode:0x110000010021 when starting a mission.

IvanOrfanidi, 2022-09-03:
Good day.

Tell me please.
At the start of the mission, we get an error 0x110000010021, but the drone continues to fly.
But at the last point, the drone stops.
```
[4275475.992]ERRORLOG/1 @ startWaypointV2Mission, L2371: Start waypoint v2 mission ErrorCode:0x110000010021
[4275475.993]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
 [4275475.993]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : waypoint plan has started when received go command.
[4275475.993]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : none
```
Could you tell me what ```ErrorCode:0x110000010021``` is and how to fix it?

Firmware version: drone: 04.01.0000 and RC: 4.01.0101.
Thanks for any help.

dji-dev, 2022-09-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/70163'>Zendesk ticket #70163</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The error message is that the drone has received repeate command . please check the code.
 [image]

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/Sp0FpEh8Zdduw9FLaYuS5tVO8' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-09-05:
Thanks for the help. We will watch and find out.

closed 2022-09-05 by IvanOrfanidi