dji-sdk/Onboard-SDK #896: Error linking OSDK

davidradakovits, 2022-09-09:
Make returns "undefined reference to `libusb_init' " among other libusb related errors allthough pkg is present on GCC: (Raspbian 8.3.0-6+rpi1) 8.3.0.

dji-dev, 2022-09-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/70426'>Zendesk ticket #70426</a>:<br><br>sudo apt-get install libusb-1.0-0-dev
<br><p><em><small>°°°</small></em></p>

davidradakovits, 2022-09-09:
The lib is present on the system: pkg-config --libs libusb returns -lusb. But actually linking against libusb with the flag -lusb doesn't work, but using the flag -lusb-1.0 works. Since the linker is called with -lusb in the project the build fails. How to change that?