dji-sdk/Onboard-SDK #902: 关于焦距问题

linssswww, 2022-11-16:
如何得到相机的实际的焦距？ 
ErrorCode::ErrorCodeType CameraManager::getOpticalZoomFactorSync(PayloadIndexType index, float &factor, int timeout)
这个函数中也只返回了factor参数，factor参数如何跟实际焦距关联起来？

linssswww, 2022-11-16:
相机的型号是H20N

dji-dev, 2022-11-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73563'>Zendesk ticket #73563</a>:<br><br>OSDK的变焦是在接口中获取到后算的。
 [image]<br><a href='https://djisdksupport.zendesk.com/attachments/token/nZZDMy39X0njp0eiYRkbI6DZl' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

linssswww, 2022-11-16:
> Agent comment from kyle.cai in [Zendesk ticket #73563](https://djisdksupport.zendesk.com/agent/tickets/73563):OSDK的变焦是在接口中获取到后算的。 [image][image.png](https://djisdksupport.zendesk.com/attachments/token/nZZDMy39X0njp0eiYRkbI6DZl)
> 
> _°°°_
那这里的lensInfo.data.min_focus_length 指的是变焦相机等效焦距还是实际焦距？

linssswww, 2022-11-16:
是这样理解嘛？ 实际的焦距 = factor * 最小的实际焦距