dji-sdk/Onboard-SDK #903: Activate API control in Assistant

gendebeat, 2022-11-23:
Hello, i want to enable API control in Assistant2-Enterprise with a M300RTK
but all fields are empy. If i check API Enable, is not save 
and baudrate have not field to select speed. 
Is someone with the same problem?

[https://ibb.co/2SrsbcZ](url)

dji-dev, 2022-11-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73922'>Zendesk ticket #73922</a>:<br><br>Because M300 newly supports PSDK 3.x (integrated with the original OSDK function), it is no longer necessary to set OSDK through DJI Assistant2, so the new version of DJI Assistant2 has changed. We have already reported to the relevant departments that the DJI Assistant2 needs to be compatible with the old OSDK4.x version.Please don't upgrade the DJI  Assistant2 now.

If you do not have a backup version of DJI Assistant2 version 2.1.3, please go to the link below to download it.
https://pan-sec.djicorp.com/s/FTYQw3LFNX3MrZ9
password：dji123

This link is valid for a period of time. If the link is invalid, please send an email to dev@dji.com. We have asked the relevant departments of DJI Assistant2 to be compatible with OSDK 4.x as soon as possible.
<br><p><em><small>°°°</small></em></p>

gendebeat, 2022-11-24:
ok thanks, now it working after spend hours fighting with version numbers.

gendebeat, 2022-11-24:
sorry maybe its not here the site for ask more questions. Now i have UART (ttyUSB0 ttl converter) and USB (ttyACM0) conect to OSDK port.
I use OSDK-ROS and i cant now connect to drone and receive the topics (telemetry). 
But maybe ttyACM0 is not working correct, because i cant receive image data in the camera topics and the log sais:
[image]
@ AdvancedSensing, L108: Failed to initialize USB Bulk Linker channel for liveview!
[2837127.288]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!

and after:
[2837127.289]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.

but i dont have image in camera topics and if i try compiled examples i dont have stream of camera. 
some forums sujest add SUBSYSTEM=="usb", ATTRS{idVendor}=="2ca3", MODE="0666"

thanks

dji-dev, 2022-11-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73922'>Zendesk ticket #73922</a>:<br><br>OSDK will use the USB ACM channel and BULK channel. The video stream goes through the bulk channel. The error message indicates that the bulk initialization failed, which is usually caused by permissions. Therefore, adding a USB node as suggested can solve this problem.
 [image]
Advanced sensing sample needs to use sudo to run（https://sdk-forum.dji.net/hc/en-us/articles/1500002017981）<br><a href='https://djisdksupport.zendesk.com/attachments/token/jPNej7Ni4rPbzK7YHVHHg1n7m' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>