dji-sdk/Onboard-SDK #909: RPI3 Won't Compile Telemetry Example 4.1.0

MasterVols, 2023-02-03:
Here is the make file:
`# *  @Copyright (c) 2016-2017 DJI
# *
# * Permission is hereby granted, free of charge, to any person obtaining a copy
# * of this software and associated documentation files (the "Software"), to deal
# * in the Software without restriction, including without limitation the rights
# * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# * copies of the Software, and to permit persons to whom the Software is
# * furnished to do so, subject to the following conditions:
# *
# * The above copyright notice and this permission notice shall be included in
# * all copies or substantial portions of the Software.
# *
# * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
#

cmake_minimum_required(VERSION 2.8)
project(djiosdk-telemetry-sample)

set(ARCH armv8)

# Compiler flags: link with pthread and enable C++11 support
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -g -O0")

# Tell the project where osdk-core is located
if(NOT ONBOARDSDK_SOURCE)
	set(ONBOARDSDK_SOURCE "/home/toor/Desktop/Onboard-SDK-master/osdk-core")
endif()
#set(CURRENT_CMAKE_MODULE_PATH ${ONBOARDSDK_SOURCE}/cmake-modules)

# Specify locations for osdk-core headers
include_directories(${ONBOARDSDK_SOURCE}/api/inc)
include_directories(${ONBOARDSDK_SOURCE}/api/src)
include_directories(${ONBOARDSDK_SOURCE}/utility/inc)
include_directories(${ONBOARDSDK_SOURCE}/utility/src)
include_directories(${ONBOARDSDK_SOURCE}/hal/inc)
include_directories(${ONBOARDSDK_SOURCE}/hal/src)
include_directories(${ONBOARDSDK_SOURCE}/protocol/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/linux/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/linux/src)
include_directories(${ONBOARDSDK_SOURCE}/linker/${ARCH}/inc)
include_directories(${ONBOARDSDK_SOURCE}/../sample/platform/linux/hal)
include_directories(${ONBOARDSDK_SOURCE}/../sample/platform/linux/hal/hotplug)
include_directories(${ONBOARDSDK_SOURCE}/../sample/platform/linux/osal)
#include_directories(include /usr/local/include)
#include_directories(include /usr/local/lib)
include_directories(${ONBOARDSDK_SOURCE}/../sample/core/inc)
include_directories(${ONBOARDSDK_SOURCE}/../sample/core/src)
include_directories(${ONBOARDSDK_SOURCE}/logger/inc)
include_directories(${ONBOARDSDK_SOURCE}/logger/src)
include_directories(${ONBOARDSDK_SOURCE}/platform/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/src)
include_directories(${ONBOARDSDK_SOURCE}/modules/inc/payload)
include_directories(${ONBOARDSDK_SOURCE}/modules/src/payload)
include_directories(${ONBOARDSDK_SOURCE}/modules/inc/filemgr)
include_directories(${ONBOARDSDK_SOURCE}/modules/src/filemgr)
include_directories(${ONBOARDSDK_SOURCE}/modules/inc/flight)
include_directories(${ONBOARDSDK_SOURCE}/modules/src/flight)
include_directories(${ONBOARDSDK_SOURCE}/modules/inc/mop)
include_directories(${ONBOARDSDK_SOURCE}/modules/src/mop)
include_directories(${ONBOARDSDK_SOURCE}/modules/inc/firewall)
include_directories(${ONBOARDSDK_SOURCE}/modules/src/firewall)
include_directories(${ONBOARDSDK_SOURCE}/api/inc/dji_vehicle)
include_directories(${ONBOARDSDK_SOURCE}/api/src/dji_vehicle)

# User-code related project files
[... 357 lines cut ...]
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:347: undefined reference to `DJI::OSDK::DataSubscription::freeMSG()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:352: undefined reference to `DJI::OSDK::Log::getErrorLogState()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:352: undefined reference to `DJI::OSDK::Log::title(int, char const*, char const*, int)'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:354: undefined reference to `DJI::OSDK::DataSubscription::freeMSG()'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/telemetry_sample.cpp.o: in function `DJI::OSDK::Telemetry::TypeMap<(DJI::OSDK::Telemetry::TopicName)8>::type DJI::OSDK::DataSubscription::getValue<(DJI::OSDK::Telemetry::TopicName)8>()':
/home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:341: undefined reference to `DJI::OSDK::Telemetry::TopicDataBase'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:341: undefined reference to `DJI::OSDK::Telemetry::TopicDataBase'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:343: undefined reference to `DJI::OSDK::DataSubscription::lockMSG()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:347: undefined reference to `DJI::OSDK::DataSubscription::freeMSG()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:352: undefined reference to `DJI::OSDK::Log::getErrorLogState()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:352: undefined reference to `DJI::OSDK::Log::title(int, char const*, char const*, int)'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:354: undefined reference to `DJI::OSDK::DataSubscription::freeMSG()'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/telemetry_sample.cpp.o: in function `DJI::OSDK::Telemetry::TypeMap<(DJI::OSDK::Telemetry::TopicName)9>::type DJI::OSDK::DataSubscription::getValue<(DJI::OSDK::Telemetry::TopicName)9>()':
/home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:341: undefined reference to `DJI::OSDK::Telemetry::TopicDataBase'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:341: undefined reference to `DJI::OSDK::Telemetry::TopicDataBase'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:343: undefined reference to `DJI::OSDK::DataSubscription::lockMSG()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:347: undefined reference to `DJI::OSDK::DataSubscription::freeMSG()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:352: undefined reference to `DJI::OSDK::Log::getErrorLogState()'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:352: undefined reference to `DJI::OSDK::Log::title(int, char const*, char const*, int)'
/usr/bin/ld: /home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:354: undefined reference to `DJI::OSDK::DataSubscription::freeMSG()'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/telemetry_sample.cpp.o: in function `DJI::OSDK::Telemetry::TypeMap<(DJI::OSDK::Telemetry::TopicName)10>::type DJI::OSDK::DataSubscription::getValue<(DJI::OSDK::Telemetry::TopicName)10>()':
/home/toor/Desktop/Onboard-SDK-master/osdk-core/api/inc/dji_subscription.hpp:341: undefined reference to `DJI::OSDK::Telemetry::TopicDataBase
`

I will be more than happy to provide extra details if needed!

durka, 2023-02-15:
All the samples are broken since 4.1.0. I raised this in #804 but DJI doesn't seem to care. Since no help for this issue appears to be forthcoming, I recommend checking out the latest working tag `4.0.1` if you want to use the samples.

MasterVols, 2023-02-17:
I ended up doing this with a rpi3 running GCC version 8.3.0. Compiling the samples eventually worked for both version 4.1 and 4.0 but I am now having another error; 

[1633358.274]ERRORLOG/1 @ getDroneVersion, L1676: Drone version not obtained! Please do not proceed.
Possible reasons:
        Serial port connection:
                * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
                * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
                * TX and RX pins are invert

I am apparently not the only one having issues with this however.

durka, 2023-02-18:
I am very familiar with that error message 😅 

The three things they suggest are actually the most common cause in my experience. Check the first two settings on the "SDK" screen in DJI Assistant. The baud rate must match your UserConfig.txt file. Also, we've found you may have to enable Timestamp at 10 Hz for it to connect.

Also, power cycle the drone after applying the settings, sometimes that's required for them to "take" 🤷 

If the settings are already correct, make sure you are plugged into the correct pins (check the user manual to see which are the OSDK pins) and try swapping TX and RX.

All that said, I do currently have an M210 with all of the above checked and it still won't connect, so faulty hardware does exist. But that's rare among hundreds of products we have shipped.