dji-sdk/Onboard-SDK #912: OSDK Setup error (running samples): Failed to Initialize Linker Channel

Shivam-Kataria, 2023-02-27:
https://developer.dji.com/onboard-sdk/documentation/introduction/homepage.html

I have been following the guide linked above. Under "running the sample", in the linux section, when I execute the command "./djiosdk-flightcontrol-sample UserConfig.txt UserConfig.txt", I get errors (see attachment of the image from the terminal). I have also attached my wiring setup as well (OSDK expansion module connects to DJI M300, onboard computer is powered on, and USB goes from the OSDK expansion module to the onboard computer). Please let me know what I am doing wrong.
[image]
[image]

dji-dev, 2023-02-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/78150'>Zendesk ticket #78150</a>:<br><br>Connect the serial port（TTL）.
 [image]<br><a href='https://djisdksupport.zendesk.com/attachments/token/QKURmqIYnDQ6UqzrJ1h5SHD2s' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

Shivam-Kataria, 2023-02-28:
I now have dual USB connected. When I run the same command while connected to M300, I get the following:
```
Read App ID
User Configuration read successfully.

[2169930.384]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2169930.384]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8C00B4GXQ
[2169930.384]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.18.7
[2169930.384]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2169930.384]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2169931.385]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[2169931.584]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[2169931.584]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2169932.847]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[2169932.847]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[2169933.848]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[2169933.855]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[2169933.855]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[2169933.857]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2169933.857]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[2169933.860]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2169933.860]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[2169933.861]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[2169933.861]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[2169934.848]STATUS/1 @ firewallTask, L112: firewall task created ...
[2169934.884]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2169934.884]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8C00B4GXQ
[2169934.884]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.18.7
[2169934.896]ERRORLOG/1 @ AdvancedSensing, L108: Failed to initialize USB Bulk Linker channel for liveview!
[2169934.896]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!
[2169934.897]ERRORLOG/1 @ AdvancedSensing, L123: Failed to initialize USB Bulk Linker channel for perception!
[2169934.897]ERRORLOG/1 @ AdvancedSensing, L130: Failed to create task for advanced sensing!
[2169934.897]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[2169934.897]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[2169935.697]STATUS/1 @ init, L254: Start advanced sensing initalization
[2169935.697]STATUS/1 @ activate, L1329: version 0x3041207

[2169936.519]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[2169936.721]STATUS/1 @ activate, L1360: Retry to activate again ..
[2169937.536]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[2169937.737]STATUS/1 @ activate, L1360: Retry to activate again ..
[2169938.552]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[2169938.753]STATUS/1 @ activate, L1360: Retry to activate again ..
[2169938.769]STATUS/1 @ getErrorCodeMessage, L744: activate
[2169938.769]STATUS/1 @ getCMDSetActivationMSG, L794: NEW_DEVICE_ERROR

[2169938.769]ERRORLOG/1 @ activate, L1380: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

[2169938.769]ERRORLOG/1 @ activate, L1387: Solutions for NEW_DEVICE_ERROR:
* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
* If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
For different aircraft, the App and the version of DJI Ass
[2169938.769]STATUS/1 @ getErrorCodeMessage, L744: initVehicle
[2169938.769]STATUS/1 @ getCMDSetActivationMSG, L794: NEW_DEVICE_ERROR

[2169938.801]STATUS/1 @ getErrorCodeMessage, L744: verify
[2169938.801]STATUS/1 @ getCMDSetSubscribeMSG, L828: SUBSCRIBER_ILLEGAL_DATA_LENGTH

[2169938.817]STATUS/1 @ getErrorCodeMessage, L744: reset
[2169938.817]STATUS/1 @ getCMDSetSubscribeMSG, L828: SUBSCRIBER_ILLEGAL_DATA_LENGTH

[2169939.229]STATUS/1 @ ~AdvancedSensing, L186: Destroy task for M300's liveview!
Segmentation fault (core dumped)
```
What am I doing wrong?

I tried googling the issue and a github issue said to add the DJIDevice.rules file in /etc/udev/rules.d with the same text that the documentation mentioned which I have already done but the same output is there.

Is the issue something to do with camera or is it with the usb connection?
Because, I did compile the code with cmake -DADVANCED_SENSING=ON however I have no SDK components attached right now

dji-dev, 2023-03-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/78150'>Zendesk ticket #78150</a>:<br><br>1、Add the DJIDevice.rules file
2、Connect DJI Assistant2 for the first time to run OSDK to activate the APPID/KEY.
<br><p><em><small>°°°</small></em></p>

Shivam-Kataria, 2023-03-01:
1) I do already have the file that contains: `SUBSYSTEM=="usb", ATTRS {idVendor} =="2ca3", MODE ="0666"`

2) I followed the link you provided, and after that, when I run it, the terminal outputs the following:
```
Read App ID
User Configuration read successfully. 

[2655087.936]ERRORLOG/1 @ initVehicle, L222: Failed to initialize ACM Linker channel!
[2655087.998]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2655087.998]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8C00B4GXQ
[2655087.998]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.18.7
[2655087.998]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2655087.998]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2655088.998]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[2655089.198]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[2655089.199]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2655089.199]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[2655089.199]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[2655089.199]STATUS/1 @ firewallTask, L112: firewall task created ...
[2655089.199]STATUS/1 @ activate, L1329: version 0x3041207

[2655089.219]STATUS/1 @ activate, L1367: Activation successful

| Available commands:                                            |
| [a] Monitored Takeoff + Landing                                |
| [b] Monitored Takeoff + Position Control + Landing             |
| [c] Monitored Takeoff + Position Control + Force Landing Avoid Ground  |
| [d] Monitored Takeoff + Velocity Control + Landing |
```

The `@ initVehicle, L222: Failed to initialize ACM Linker channel!` is still there, and the `USB is not plugged or initialized successfully. Advacned-Sensing will not run.` I have a dual USB connected.

closed 2023-03-03 by Shivam-Kataria

Barathwaja, 2023-03-06:
Hi @Shivam-Kataria ,

I'm also facing same issue, but am not using Manifold-2. Instead connected directly to Ubuntu Computer with OSDK cable and USB-C cable. Could you please let me know exact steps that helped to solve this?

Thanks in advance.

Shivam-Kataria, 2023-03-06:
Hi @Barathwaja, I was not using manifold either. You need a dual USB connection between the onboard computer and the OSDK expansion module. Like the one specified by @dji-dev in image in the chat above. The expansion module is connected to the the M300 using its usbc cable

Barathwaja, 2023-03-06:
This is how I have setup.

1- OSDK cable which connected to Computer
2- USB-C another connected to computer
So overall 2 connections 

[image]

Please correct me if am wrong.

Shivam-Kataria, 2023-03-07:
@Barathwaja I am not sure if that will work or not. I purchased https://store.dji.com/ca/product/matrice-300-rtk-osdk-expansion-module?vid=93051&from=pc-cart-title

The way I was wired up was the same as what dji-dev showed with the dual usb connection, one usb and the other ttl

In other words, these are the wires connected:
- USBC cable from the expansion module to the drone
- TTL usb cable from the pins of the expansion module to the usb port on the onboard computer
- usb cable from the expansion module to the onboard computer

On this page, it shows a way to make your own expansion module instead of buying it https://forum.dji.com/thread-219723-1-1.html but I have personally not tried it.

Barathwaja, 2023-03-07:
Thanks yeah I think I got it now.

Below is the setup and do we need to power XT30?

[image]

Shivam-Kataria, 2023-03-07:
The XT30 is output power, so you can wire it to your onboard computer and it would use the drone batteries to power your computer. I believe it is 24V.

Barathwaja, 2023-03-07:
Hi @Shivam-Kataria ,

Understood. I'm getting this below error. tried to change the baudrate & TX + RX pins still same issue.

1- black ground, 2- white RX into USB port, and 3 -green TX out of the USB port. Should i change order anything?

Thanks in advance.

[image]

Shivam-Kataria, 2023-03-07:
Make sure you have enabled api control in dji assistant 2

The baudrate in your config.txt file should match with the baudrate in dji assistant 2

The order of pins is shown in this image in the osdk expansion module manual in column 3 from top to bottom
[image]

In the /etc/udev/rules.d directory, you should have a file that contains the DJIDevice.rules file with the following text:
`SUBSYSTEM=="usb", ATTRS {idVendor} =="2ca3", MODE ="0666"` as specified in the documentation

the top error "failed to initalize ACM linker channel" means that the USB ACM port is not recognized, After the dual A USB is connected correctly, check the recognized ACM port through "ls /dev | grep ttyACM*. and then the ACM name in the OSDK configuration file needs to be consistent with the recognized name.

Barathwaja, 2023-03-07:
Yeah I did all of these but still same issue. Also, I connected the DJI Assistant2 but how does the activation works?

Barathwaja, 2023-03-09:
Thank you @Shivam-Kataria . I had to use different UART TTL stuff (FT232) to make it work.