dji-sdk/Onboard-SDK #914: Starting Development Questions

Shivam-Kataria, 2023-03-09:
Hi,

I was reading over the DJI OSDK documentation but I have not been able to figure out how to develop with OSDK yet. As in, lets say I want to write custom code to make the dji do some basic things such as take off, move forward by some meters and land. How do I go on about doing that with the osdk? I am not sure how the setup of the cpp project would look like to integrate working with dji osdk (I am relatively new to cpp).

Also, are there any tutorials on dji osdk development other than the documentation itself. I have not been able to find anything anywhere (on YouTube for example).

Barathwaja, 2023-04-01:
Hi @Shivam-Kataria ,

They stopped upgrading codes for OSDK and their suggestion is to move forward with PSDK. Cause I ran into lot of dependency library issues with OSDK in Ubuntu so go with PSDK.

You can use OSDK port and run PSDK module code and I guess they are going to make PSDK as primary / one-stop library.

licsber, 2023-04-06:
> Hi @Shivam-Kataria ,
>
> They stopped upgrading codes for OSDK and their suggestion is to move forward with PSDK. Cause I ran into lot of dependency library issues with OSDK in Ubuntu so go with PSDK. [...]

haha.

I recently just start learning how to use the OSDK to control my M300-RTK, struggling for compile this code. DJI official suggest Ubuntu 16.04 with OpenCV3 , but my dev platform must be Ubuntu 20.04 with OpenCV4.

Is there any tutorials for use PSDK on OSDK port? I have OSDK expansion module, does just connecting OSDK port work? or i should buy the PSDK suites?

Barathwaja, 2023-04-06:
Hi @licsber ,

Yes. DJI won't upgrade code so we cannot use the OSDK and run on newer version of OS. Switch to [PSDK module](https://github.com/dji-sdk/Payload-SDK) and run the code connecting the Expansion Board on the OSDK Port.

Unfortunately the PSDK english version is not clearly explained about how they structured the project. So I found a [Chinese version](https://sdk-forum.dji.net/hc/zh-cn/sections/4591593564441-PSDK3-x-OSDK-PSDK-%E7%B3%BB%E5%88%97%E6%96%87%E6%A1%A3) explained clearly you can use google translate them.

@dji-kyle @dji-dev I have already raised an incident. So please start working on translating for the them.

licsber, 2023-04-07:
> Hi @licsber ,
>
> Yes. DJI won't upgrade code so we cannot use the OSDK and run on newer version of OS. Switch to [PSDK module](https://github.com/dji-sdk/Payload-SDK) and run the code connecting the [...]

Thanks, it very helpful to me.

Fortunately, I'm Chinese, [smile], Now i can learn the new one.

Very ashamed, for i even didn't find this document.

closed 2023-10-12 by Shivam-Kataria