dji-sdk/Onboard-SDK #918: ERRORLOG/1 @ initVehicle, L215: Failed to initialize linker channel. Vehicle not initialized

fabioCarfi95, 2023-06-13:
Hello everyone,

I'm trying to customize the OSDK 4.1.0.
I followed all the passages explained in the [run the sample page](https://developer.dji.com/onboard-sdk/documentation/quickstart/run-the-sample.html) but, when I execute the point 2 of chapter _''Execute the Sample program''_, it gives me the error written in the title.
Thinking that I could have failed something on the changes applied, I downloaded an OSDK copy and followed the steps again without any changes but it gives me the same error.

Can someone help me?

dji-dev, 2023-06-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/85004'>Zendesk ticket #85004</a>:<br><br>“Failed to initialize linker channel.”
It means that the UART port initialization failed. it is need to check that the UART port name configured on the OSDK side and the UART port actually used are correct.
<br><p><em><small>°°°</small></em></p>

fabioCarfi95, 2023-06-14:
I checked the UART port name and give them all the permissions as specified in the first point of _Compile_ Chapter in the [Run the Sample page](https://developer.dji.com/onboard-sdk/documentation/quickstart/run-the-sample.html).
Unfortunately, I continue having the same problem. What can be more other operations to resolve this problem?

dji-dev, 2023-06-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/85004'>Zendesk ticket #85004</a>:<br><br>This linker corresponds to the UART port of the onboard device connected to the TTL of the drone, not ttyACM0, which is usually the ACM device recognized after the dual A USB is connected to the drone.
https://developer.dji.com/document/1bb1b2dc-9694-4d6b-ad43-fa69c5a10400<br><a href='https://djisdksupport.zendesk.com/attachments/token/wGoa5FYScJI0DfjxvHypTDCbY' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

fabioCarfi95, 2023-06-15:
So, for using the OSDK I HAVE TO use only the UART cable instead of the dual A USB one?
For what I read in other tutorial, I was pretty sure that the USB is the right one