dji-sdk/Onboard-SDK #926: connect a DJI Matrice 350 RTK to a Nvidia Jetson AGX Orin

LuisOrtizF, 2023-09-04:
Hi,

How can I connect a DJI Matrice 350 RTK to a Nvidia Jetson AGX Orin?

dji-dev, 2023-09-05:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90365'>Zendesk ticket #90365</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

You can refer to the following article to connect and set up the environment. Due to system issues, I switched languages abnormally. You may need to open this link before translating.
https://sdk-forum.dji.net/hc/zh-cn/articles/4612590927769?source=search&auth_token=eyJhbGciOiJIUzI1NiJ9.eyJhY2NvdW50X2lkIjo5Nzg4MTcsInVzZXJfaWQiOjIwMzk2ODQ5MTA3MjI1LCJ 0aWNrZXRfaWQiOjg3NjAwLCJjaGFubmVsX2lkIjo2MywidHlwZSI6IlNFQVJDSCIsImV4cCI6MTY5Mjg2MDIzOX0.7VNX7AvlKZetuwWGNHr1HZ2d8IYuHBpQWioAjBDn8Zc

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

SamDSchofield, 2023-09-12:
Hi @LuisOrtizF, did you manage to get the OSDK working with your M350? I saw the [other issue](https://github.com/dji-sdk/Onboard-SDK/issues/925) which conflicts with the advice given here I think. I'm in the process of upgrading from an m300 to an m350 and am unsure if it isn't working because I've configured something wrong or if the 350 just isn't supported.

dji-dev, 2023-09-12:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90365'>Zendesk ticket #90365</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

If you want to use the OSDK development kit, sorry, M350 can no longer be developed using OSDK. You can use the latest PSDK development kit to continue development, which contains the original functions of OSDK.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>