dji-sdk/Onboard-SDK #927: Question to start development

andreamanera, 2023-09-13:
Good morning, I am Andrea a student who is currently developing his master thesis in the field of robotics engineering, and as part of the project I need to connect a Raspberry pi 4 model B, which should operate as the onboard computer for a Matrice 210 RTK V2, as I'm setting up the Raspberry pi, and I would like to install Linux (Ubuntu) as operating system, I was wondering if there is a specific version of the OS that is preferred among others. To give some context the idea is to develop a system to manage a fleet of UAVs using a master slave architecture with UAVs communicating through a mesh wifi network, the Matrice 210 should operate as master, receiving telemetry data from the other UAVs (also the other drones will operate with the onboard SDK) and sending them flight plans.

dji-dev, 2023-09-14:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90951'>Zendesk ticket #90951</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

You can use UBUNTU 16.04 or 18.04.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

andreamanera, 2023-09-15:
okay, thank you for your answer, so if I try to use Ubuntu 20.04 should I have compatibility problems with the OSDK?

dji-dev, 2023-09-15:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90951'>Zendesk ticket #90951</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

We have not tested on 20.0. Other developers have also used this system version, but we may not have a lot of information to provide you with regards to troubleshooting problems on this system.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

andreamanera, 2023-09-22:
Good morning, I don't know if it is okay to post more on this issue, if it's not feel free to remove my question and I'll open a new issue, anyway I was able to correctly setup the Raspberry 4b with Ubuntu 18.0.4 and to correctly connect it to the drone to run the sample code, now I have a question that is more specific, I would like to access with the onboard computer the coordinate of the area that I'm going to call working area, which is defined on DJI pilot 2 when planning a new mission (the blue area of which you can change the coordinates in the mission planning screen to be clear) do you think that this is possible? And if so do you know where I should look in the OSDK APIs?

dji-dev, 2023-09-22:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90951'>Zendesk ticket #90951</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Sorry, it may be a problem with the translation software. I don't quite understand what you mean. If possible, could you describe it more specifically or record some videos? I think there should be no ambiguity in communication like this.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

andreamanera, 2023-09-22:
I have no video of what I'm trying to ask at the moment anyway I'll post a picture and I’ll try to be clearer. Looking at the picture that I have attached you can see that there is a light blue area that is what I was referring to as "working area" and it's defined when planning a new mission, I would like to know if coordinates of the vertices of this area are "sent" from the RC to the drone (Matrice 210 in my case) and if so I would like to know if there exist an API which give me the possibility to get this coordinates from the on-board computer. I hope that you can get my question, if no tell me and I will try to explain myself again
[image]

dji-dev, 2023-09-25:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90951'>Zendesk ticket #90951</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

I understand what you mean. I confirmed this problem with the R&D team. I'm sorry, there is no such TOPIC in O/PSDK at present, and the information in this workspace does not interact with PSDK.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>