dji-sdk/Onboard-SDK #931: [M300 RTK] Failed to upload mission v2

btsatoukzn, 2023-11-16:
hi

When uploading a mission, we encountered the phenomenon described below. We also found a solution. Please confirm whether the solution is appropriate.

・Events that occur
For example, register 10 WP missions. At this time, set WP7 and WP8 to the same latitude, longitude, and altitude. Too close error message is displayed.
Next, register WP3 normal missions. However, mission registration for 3 WPs fails with the following error.
"the start index is greater than end index of upload wps"

・Cause of this event
Static variables in the API source"dji_waypoint_v2.cpp" appear not to be initialized.

·Solution
Modify the static variable so that it is initialized when the mission is uploaded.

thank you.

btsatoukzn, 2023-11-16:
78 raw
`bool missionEncode(const std::vector<WaypointV2Internal> &mission, uint8_t *pushPtr,
                   uint16_t &len) {
  static uint16_t startIndex = 0;
  if (mission.empty()) {
    pushPtr = nullptr;
    len = 0;
    startIndex = 0;
    return true;
  }

  bool finished = false;
  uint16_t tempTotalLen = 0;
  //static uint16_t startIndex = 0;
  uint16_t endIndex = 0;
  uint8_t *tempPtr = pushPtr;

  /*! {{startIndex, endIndex, waypoint1, waypoint2,...},{startIndex, endIndex,
   * waypoint1, waypoint2,...}, ....*/
  elementEncode<uint16_t>(startIndex, tempTotalLen, tempPtr);
  uint8_t *tempTempPtr = tempPtr;

`

around 569 raw
`ErrorCode::ErrorCodeType WaypointV2MissionOperator::uploadMission(
  int timeout) {
  
  std::vector<WaypointV2Internal> mission_empty;
  uint16_t empty_push = 0;
  missionEncode(mission_empty, nullptr, empty_push);

  std::vector<WaypointV2Internal> mission = transformMission2MisssionInternal(this->missionV2);

`

dji-dev, 2023-11-20:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/95170'>Zendesk ticket #95170</a>:<br><br>This may not be the problem, startIndex is initialized to 0 in the next  code.

    bool finished = false; uint16_t tempTotalLen = 0; static uint16_t startIndex = 0; uint16_t endIndex = 0; uint8_t *tempPtr = pushPtr;

You can see startIndex and endIndex in the log print in missionEncode method.
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2023-11-20:
thanks for your reply.

Since it's a static field, we don't think it will be initialized the second time.

thank you.

dji-dev, 2023-11-20:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/95170'>Zendesk ticket #95170</a>:<br><br>Sorry, I misunderstood you before. If you upload the file twice, your modifications are indeed OK.
<br><p><em><small>°°°</small></em></p>