dji-sdk/Onboard-SDK #932: Questions on Tracking Module

Daniel-xsy, 2023-12-05:
Hi,

May I have two questions:

1. I'm wondering if the Onboard-SDK can only be applied to Matrice series, which means I can't develop new features on other drones (e.g., the newest Mini 4 Pro).
2. I noticed the demo here: https://developer.dji.com/cn/onboard-sdk/documentation/sample-doc/advanced-sensing-target-tracking.html shows target tracking sample with only the gimbal rotated. I'm wondering if it's possible to build ActiveTrack using the SDK to send the tracking results to the control loop. If so, then can we implement a customized tracking algorithm in ActiveTrack?

Thanks so much.

Daniel-xsy, 2023-12-05:
Thanks for your reply.

For the second question, I mean can we implement ActiveTrack using the Onboard-SDK, which uses vision tracking to follow a target.

Daniel-xsy, 2023-12-06:
Also, is anyway to get the tracking results (e.g., `(x, y, w, h)`) of the ActiveTrack deployed on products? Thanks

dji-dev, 2023-12-06:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/96416'>Zendesk ticket #96416</a>:<br><br>I apologize for the inconvenience. Onboard-SDK can only obtain visual information. If you need to implement the ActiveTrack function, you need to develop and implement it.
<br><p><em><small>°°°</small></em></p>

closed 2023-12-19 by Daniel-xsy