dji-sdk/Onboard-SDK #935: Build error in the Yocto system

bchoineubility, 2024-04-05:
Hi,

I am trying to build OSDK in the Yocto, and I faced build error which closed already by #889 .
But, I can not use gcc 5.4 with Yocto.

Error message :

```
osdk-core/modules/inc/filemgr/impl/mmap_file_buffer.hpp:23:15: error: field 'currentLogFilePath' has incomplete type 'std::string' {aka 'std::__cxx11::basic_string<char>'}
```

By the way, I add `#include <string>` to mmap_file_buffer.hpp and disappeared error. Here is my patch file.

Can you check it available?

[0001-Remove-compile-error-by-incompatible-string-type.patch](https://github.com/dji-sdk/Onboard-SDK/files/14880044/0001-Remove-compile-error-by-incompatible-string-type.patch)

dji-dev, 2024-04-07:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/104293'>Zendesk ticket #104293</a>:<br><br>Dear bchoineubility,

Greetings. Thank you for reaching out to DJI Innovations.

We appreciate your patience. We apologize for any misunderstanding that may have occurred due to translation software. Are you requesting us to replace this patch and then proceed with the OSDK functionality testing? Or are you referring to testing a specific feature?

Thank you for your support of DJI products! We wish you all the best.

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

bchoineubility, 2024-04-07:
Hi @dji-dev 

I am brand new this sdk. 
I hope to know my patch correct or not and check out reason for compile error. If it is gcc version problem, I should tried to use gcc 5.4 I think.

It's your freedom to use this patch or not.

BR,
Mark

dji-dev, 2024-04-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/104293'>Zendesk ticket #104293</a>:<br><br>Dear bchoineubility,

Greetings! Thank you for reaching out to DJI - Da-Jiang Innovations.

We appreciate the information you have provided. We will attempt to see if it is usable. Do you currently have any other doubts or questions? Thank you once again!

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>