dji-sdk/Onboard-SDK #938: DJI M300 Inertial and Drone Parameters of Config File

adeelahsaanawan, 2024-07-26:
We are designing optimal guidance, navigation, and control algorithms and aim to test those algorithms on the DJI M300 drone. For our research purposes, we need drone inertial parameters, like Inertia Tensor Matrix, mass, drag coefficient, motor thrust coefficient, motor torque coefficient, motor time constant, arm length, etc. Could anyone guide me on how I can get these parameters? Or can someone refer me to any drone configuration file from DJI OSDK since we use DJI OSDK to implement the algorithms on the manifold 2G?

dji-dev, 2024-07-29:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113019'>Zendesk ticket #113019</a>:<br><br>Dear adeelahsaanawan,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

We appreciate your patience. We regret to inform you that the information you have mentioned may not fully meet your requirements. If you are looking to implement algorithmic flight control, you can first compute the fusion of the data you have collected on the onboard computer, and then control the aircraft's flight through the PSDK interface. The coefficients you mentioned, such as drag coefficient, motor thrust coefficient, and motor time length, cannot be disclosed.

For other information such as arm length, we have publicly available model drawings which you can view here: https://github.com/dji-sdk/Payload-SDK/tree/master/doc/simple_model

Additionally, you can find its basic information here: https://developer.dji.com/doc/payload-sdk-tutorial/cn/model-instruction/choose-hardware-platform.html

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

adeelahsaanawan, 2025-02-13:
Can you atleast give the inertia tensor matrix?