dji-sdk/Onboard-SDK #939: Function velocityAndYawRateCtrl() does not work when no GPS signal.

Lisiiii, 2024-07-27:
When flying indoors in the real world without GPS signal, I use velocityAndYawRateCtrl() in Onboard SDK to control the drone flight (the remote controller is in P mode) but it has no effect. However, other control functions, including those with Z-axis speed control, work normally.
Everything works fine in the simulator. I don't know if speed control can't be used without GPS, or if there is a problem with velocityAndYawRateCtrl() function.
I use OSDK version 3.9, tested on A3/N3 flight controller.

dji-dev, 2024-07-29:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113046'>Zendesk ticket #113046</a>:<br><br>Dear Lisiiii,

Greetings. We appreciate your contact with DJI Innovations.

We regret any inconvenience caused. The control method velocityandyawrateCtrl() relies on GPS, so calling it without GPS or in poor connection conditions may result in exceptions. We also do not recommend using the velocityandyawrateCtrl method for indoor control.

Thank you for your support of DJI products! We wish you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

Lisiiii, 2024-07-29:
I see. Thank you for your reply.

closed 2024-07-29 by Lisiiii