dji-sdk/Onboard-SDK #940: Advanced Sensing error when trying to get frames from the DJI Zenmuse H20T camera

Yiannis15, 2024-08-05:
Hello, when i try to get frames the camera sometimes i get this error and i cant get a frame. The only solution i found is restarting the drone. Is there a possible solution to fix this problem? I use a Jetson Xavier NX that is mounted on a DJI Matrice 300 using the onboard sdk module. Also the udev rules set up is already done.

I am attaching 2 images that might be helpful
[image]
[image]

Yiannis15, 2024-08-05:
Greetings, when i restart the drone the problem is fixed yes. But usually it appears on random times. I am using the Onboard version 4.1.0

dji-dev, 2024-08-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113734'>Zendesk ticket #113734</a>:<br><br>Greetings Yiannis15,

Thank you for contacting DJI Innovations.

We apologize for the inconvenience you have experienced. The issue you are encountering appears to be related to the USB connection. We recommend that you try replacing the USB cable to see if this resolves the problem. Additionally, it is possible that this is a bug in the OSDK. Since the OSDK is no longer being updated or maintained, this issue may not be resolved at the code level. If you are using the M300 model, we suggest that you use the PSDK for development, as the PSDK is continuously updated and maintained.

Thank you for your support of DJI products. We wish you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

Yiannis15, 2024-08-09:
Greetings,

How can we connect a Jetson Xavier NX to the DJI Matrice 300 to use the PSDK.

Best Regards.

dji-dev, 2024-08-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113734'>Zendesk ticket #113734</a>:<br><br>Greetings Yiannis15,

Thank you for contacting DJI Innovations.

You may refer to the following link:

PSDK connection of various aircraft models

We appreciate your support for DJI products and wish you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

Yiannis15, 2024-08-09:
Greetings, I dont see a link

Best Regards

dji-dev, 2024-08-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113734'>Zendesk ticket #113734</a>:<br><br>Dear Developer,

Greetings, and thank you for contacting DJI Innovations.

You may refer to the following link for information:

https://sdk-forum.dji.net/hc/en-us/articles/34834641217305-PSDK-connection-of-various-aircraft-models

We appreciate your support for DJI products and wish you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>