dji-sdk/Onboard-SDK #942: Download image failed

wuxiangchao, 2025-01-15:
when i run "./payload_download_sample", get the following error:
[image]

dji-dev, 2025-01-15:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127117'>Zendesk ticket #127117</a>:<br><br>Hello, thank you for your patience. This may be a bug in our OSDK. Unfortunately, since OSDK has stopped maintenance and updates, this problem may not be fixed. We recommend that you use PSDK for development, which we are continuing to update and maintain.
<br><p><em><small>°°°</small></em></p>

wuxiangchao, 2025-01-16:
> Agent comment from Leon in [Zendesk ticket #127117](https://djisdksupport.zendesk.com/agent/tickets/127117):Hello, thank you for your patience. This may be a bug in our OSDK. Unfortunately, since OSDK has stopped maintenance and updates, this problem may not be fixed. We recommend that you use PSDK for development, which we are continuing to update and maintain.
> 
> _°°°_

okk, thanks for your reply.