dji-sdk/Onboard-SDK #944: Activation every time I connect the Jetson Xavier NX with the Matrice 300 RTK

Y1annis, 2025-04-08:
Hello!

Every time I run my OSDK program after removing the jetson from the drone i get the NEW_DEVICE_ERROR error. I have wifi on my jetson, controller but I have to connect my Laptop using DJI Assistant 2 that is logged in into the account that has the api key to activate the OSDK program.

Anyone know why is this happening?

Thank you for your time

dji-dev, 2025-04-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138933'>Zendesk ticket #138933</a>:<br><br>Hello, what version of the M300 model and DJI Assistant 2 software are you using? Normally, OSDK only needs to be activated once.
In addition, we recommend that you use PSDK for development, because OSDK has stopped updating and maintenance, and there may be some bugs that affect use.
<br><p><em><small>°°°</small></em></p>

Y1annis, 2025-04-09:
Hello! I am using DJI Assistant Enterprise, and the m300 model is updated to the latest software

dji-dev, 2025-04-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138933'>Zendesk ticket #138933</a>:<br><br>Hello, could you please check the version number on the bottom right corner of the DJI Assistant2 software? As shown in the picture, this number indicates the version number.
In addition, could you please show us the OSDK running error?<br><a href='https://djisdksupport.zendesk.com/attachments/token/GmYaGXyGL4gIhzImILPsAkYRl' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>