dji-sdk/Onboard-SDK #945: Questions about limitations of gimbal interaction via SDK

88brewing, 2025-04-23:
Hello! I am a current graduate student using a Matrice 210 V2 with a Zenmuse gimbal for a research project. I have successfully set up and ran some code using the synchronous gimbal manager sample found within this repository.

My project relies on accurate readings from the gimbal. Right now, I can use the synchronous gimbal manager sample to read gimbal orientation with a precision of 0.1 degrees. For my project this is not sufficient. My question is if there is any known way (even if it is complex, I am looking for any feasible solutions) to **a)** read the gimbal gyroscope/accelerometer values directly, as can be done with the telemetry of the drone itself, and **b)** increase the precision of the reported gimbal orientation, preferably to at least a precision of 0.01 degrees but as accurate as possible is best.

dji-dev, 2025-04-24:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/140833'>Zendesk ticket #140833</a>:<br><br>Hello, I'm sorry, the current supported accuracy is 0.1, and there is no way to improve it further, because the data pushed from the flight control end is 0.1 degree accuracy. As this product has been around for a long time, we will no longer update the firmware to modify this. We apologize for the inconvenience caused to you.
<br><p><em><small>°°°</small></em></p>

88brewing, 2025-04-24:
> Agent comment from Leon in [Zendesk ticket #140833](https://djisdksupport.zendesk.com/agent/tickets/140833):Hello, I'm sorry, the current supported accuracy is 0.1, and there is no way to improve it further, because the data pushed from the flight control end is 0.1 degree accuracy. As this product has been around for a long time, we will no longer update the firmware to modify this. We apologize for the inconvenience caused to you.
> 
> _°°°_

Okay, I appreciate the response. What about reading values from the gimbal gyroscope directly, is there any way to make this possible?