dji-sdk/Onboard-SDK #946: Onboard-SDK4.1示例程序运行报错

AC-DB, 2025-05-01:
已经添加C/C++编译器附加选项
```shell
--exceptions --cpp11 -DFLIGHT_CONTROL_SAMPLE
```
已经通过sh脚本自动导入FreeRTOS源码

环境使用 `VScode + EIDE ` , ` ARM_Compiler_5.06u7 ` 

下面是编译器输出日志

```shell
[ INFO ] start building at 2025-05-01 15:23:10

[ TOOL ] Component: ARM Compiler 5.06 update 7 (build 960)

[ INFO ] file statistics (incremental mode)

+---------+-----------+-----------+---------------+--------+
| C Files | Cpp Files | Asm Files | Lib/Obj Files | Totals |
+---------+-----------+-----------+---------------+--------+
| 7       | 15        | 0         | 1             | 23     |
+---------+-----------+-----------+---------------+--------+

[ INFO ] start compiling (jobs: 8) ...

>> [  4%] CC '../Lib/USB_Lib/STM32_USB_HOST_Library/Class/CDC/src/usbh_cdc_core.c'
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 152: Error:  #144: a value of type "void *" cannot be used to initialize an entity of type "USBH_HOST *"
    USBH_HOST *pphost = phost;
                        ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 308: Error:  #144: a value of type "void *" cannot be used to initialize an entity of type "USBH_HOST *"
    USBH_HOST *pphost = phost;
                        ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 318: Error:  #167: argument of type "void *" is incompatible with parameter of type "USBH_HOST *"
      ClassReqStatus = CDC_GETLineCoding(pdev, phost);
                                               ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 328: Error:  #167: argument of type "void *" is incompatible with parameter of type "USBH_HOST *"
      ClassReqStatus = CDC_SETLineCoding(pdev, phost);
                                               ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 343: Error:  #167: argument of type "void *" is incompatible with parameter of type "USBH_HOST *"
      ClassReqStatus = CDC_SETControlLineState(pdev, phost);
                                                     ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 358: Error:  #167: argument of type "void *" is incompatible with parameter of type "USBH_HOST *"
                                       phost,
                                       ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c", line 385: Error:  #144: a value of type "void *" cannot be used to initialize an entity of type "USBH_HOST *"
    USBH_HOST *pphost = phost;
                        ^
.\..\Lib\USB_Lib\STM32_USB_HOST_Library\Class\CDC\src\usbh_cdc_core.c: 0 warnings, 7 errors
>> [  9%] CC '../Lib/USB_Lib/STM32_USB_HOST_Library/Core/src/usbh_stdreq.c'
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Core\src\usbh_stdreq.c", line 493: Error:  #167: argument of type "void *" is incompatible with parameter of type "std::uint8_t *"
                  pdesc = USBH_GetNextDesc((void* )pdesc, &ptr);
                                           ^
.\..\Lib\USB_Lib\STM32_USB_HOST_Library\Core\src\usbh_stdreq.c: 0 warnings, 1 error  
>> [ 13%] CC '../Lib/USB_Lib/STM32_USB_HOST_Library/Core/src/usbh_core.c'
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Core\src\usbh_core.c", line 590: Error:  #167: argument of type "const char *" is incompatible with parameter of type "void *"   
        phost->usr_cb->ManufacturerString("N/A");
                                          ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Core\src\usbh_core.c", line 611: Error:  #167: argument of type "const char *" is incompatible with parameter of type "void *"   
        phost->usr_cb->ProductString("N/A");
                                     ^
".\..\Lib\USB_Lib\STM32_USB_HOST_Library\Core\src\usbh_core.c", line 632: Error:  #167: argument of type "const char *" is incompatible with parameter of type "void *"   
        phost->usr_cb->SerialNumString("N/A");
                                       ^
.\..\Lib\USB_Lib\STM32_USB_HOST_Library\Core\src\usbh_core.c: 0 warnings, 3 errors   
>> [ 18%] CC '../Linker/osal/osdkosal_stm32.c'
".\..\Linker\osal\osdkosal_stm32.c", line 51: Error:  #167: argument of type "T_OsdkTaskHandle" is incompatible with parameter of type "TaskHandle_t *"
    result = xTaskCreate((TaskFunction_t)taskFunc, taskName, stackSize, arg, 0, *task);
                                                                                ^    
".\..\Linker\osal\osdkosal_stm32.c", line 61: Error:  #167: argument of type "T_OsdkTaskHandle" is incompatible with parameter of type "TaskHandle_t"
    vTaskDelete(task);
[... 54 lines cut ...]
                       ^
".\..\OS\FreeRTOS\tasks.c", line 3593: Error:  #513: a value of type "void *" cannot be assigned to an entity of type "TCB_t *"
                                pxTCB = listGET_OWNER_OF_HEAD_ENTRY( ( &xTasksWaitingTermination ) ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of the pointer stored and retrieved is the same. */
                                      ^
".\..\OS\FreeRTOS\tasks.c", line 3712: Error:  #513: a value of type "void *" cannot be assigned to an entity of type "TCB_t *"
                        listGET_OWNER_OF_NEXT_ENTRY( pxFirstTCB, pxList ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of the pointer stored and retrieved is the same. */  
                        ^
".\..\OS\FreeRTOS\tasks.c", line 3720: Error:  #513: a value of type "void *" cannot be assigned to an entity of type "TCB_t *"
                                listGET_OWNER_OF_NEXT_ENTRY( pxNextTCB, pxList ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of the pointer stored and retrieved is the same. */
                                ^
".\..\OS\FreeRTOS\tasks.c", line 3898: Error:  #513: a value of type "void *" cannot be assigned to an entity of type "TCB_t *"
                ( pxTCB ) = listGET_OWNER_OF_HEAD_ENTRY( pxDelayedTaskList ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of the pointer stored and retrieved is the same. */
                          ^
.\..\OS\FreeRTOS\tasks.c: 0 warnings, 10 errors
>> [ 36%] CXX '../User/Activate.cpp'

 ERROR  compilation failed at : "c:\Users\A-bcd\Desktop\Program\robot\onboard\Onboard-SDK-4.1.0_3\Onboard-SDK-4.1.0\sample\platform\STM32\OnBoardSDK_STM32\OS\FreeRTOS\tasks.c", exit code: 1
command:
  "D:\Tools\Keil_v5\ARM\ARM_Compiler_5.06u7\bin\armcc.exe" -c --apcs=interwork -I../Lib/CMSIS/inc -I../Lib/STM32F4xx_StdPeriph_Driver/inc -I../User -I../Bsp -I../../../../core/inc -I../OS/FreeRTOS/include -I../OS/FreeRTOS/portable/RVDS/ARM_CM4F -I../Linker/hal -I../Linker/osal -I../Lib/USB_Lib/STM32_USB_HOST_Library/Class/CDC/inc -I../Lib/USB_Lib/STM32_USB_HOST_Library/Core/inc -I../Lib/USB_Lib/STM32_USB_OTG_Driver/inc -I../../../../../osdk-lib/STM32/osdk-core/inc -I../../../../../osdk-core/api/inc -I../../../../../osdk-core/linker/STM32/inc -I../../../../../osdk-core/logger/inc -I../../../../../osdk-core/modules/inc/payload -I../../../../../osdk-core/modules/inc/flight -I../../../../../osdk-core/modules/inc/firewall -I../../../../../osdk-core/platform/inc -I../../../../../osdk-core/utility/inc -I.cmsis/include -IRTE/_DJI_LIB -DUSE_STDPERIPH_DRIVER -DSTM32F40_41xxx -DUSE_STM324xG_EVAL -DSTM32 -DSTM32F40x -DARM_MATH_CM4 -D__FPU_PRESENT -DUSE_USB_OTG_FS --cpu Cortex-M4.fp --li --c99 -D__MICROLIB -O0 --split_sections --gnu -W --exceptions --cpp11 -DFLIGHT_CONTROL_SAMPLE -g -o .\build\DJI_LIB\.obj\__\OS\FreeRTOS\tasks.o --no_depend_system_headers --depend .\build\DJI_LIB\.obj\__\OS\FreeRTOS\tasks.d .\..\OS\FreeRTOS\tasks.c

 ERROR  build failed !, elapsed time 0:0:0

 *  终端进程“cmd.exe /C "unify_builder -p c:\Users\A-bcd\Desktop\Program\robot\onboard\Onboard-SDK-4.1.0_3\Onboard-SDK-4.1.0\sample\platform\STM32\OnBoardSDK_STM32\Project\build\DJI_LIB\builder.params"”已终止，退出代码: 1。 
 *  终端将被任务重用，按任意键关闭。 
```

dji-dev, 2025-05-01:
Agent comment from SHENRONG.LE in <a href='https://djisdksupport.zendesk.com/agent/tickets/141641'>Zendesk ticket #141641</a>:<br><br>Dear developers,
Hello, thank you for contacting DJI Innovation.

The engineer is currently on leave and will contact you as soon as the holiday ends. Thank you for your understanding.

Thank you for your email and have a nice life!
Best Regards,
Dji innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

AC-DB, 2025-05-05:
现在我改用了
 
-  `STM32F4XX_DFP` 的版本为 `2.12.0`  
-  `ARM::CMSIS` 的版本为 `6.1.0` 

当我编译时出现下面这种错误 

```shell
Build target 'DJI_LIB'
compiling BasicMathFunctions.c...
D:\Tools\Keil_v5\ARM\ARMCLANG\include\arm_acle.h(21): error:  #35: #error directive: "ACLE intrinsics support not enabled."
  #error "ACLE intrinsics support not enabled."
.........
D:/Tools/Arm/Packs/ARM/CMSIS-DSP/1.15.0/Source/TransformFunctions/TransformFunctionsF16.c: 0 warnings, 1 error
".\Objects\osdk_stm32.axf" - 28 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:02
```

但是当我选择 `ARM::CMSIS` 的版本为 `5.9.0`  出现下面这钟的类似情况

```
compiling dji_psdk_module.cpp...
compiling ControllerFunctions.c...
..\Lib\CMSIS\inc\core_cmInstr.h(114): error:  #247: function "__REV16"  has already been defined
  __attribute__((section(".rev16_text"))) __STATIC_INLINE __ASM uint32_t __REV16(uint32_t value)
..\Lib\CMSIS\inc\core_cmInstr.h(128): error:  #247: function "__REVSH"  has already been defined
  __attribute__((section(".revsh_text"))) __STATIC_INLINE __ASM int32_t __REVSH(int32_t value)
..\Lib\CMSIS\inc\core_cmFunc.h(50): error:  #247: function "__get_CONTROL"  has already been defined
  __STATIC_INLINE uint32_t __get_CONTROL(void)
..\Lib\CMSIS\inc\core_cmFunc.h(63): error:  #247: function "__set_CONTROL"  has already been defined
  __STATIC_INLINE void __set_CONTROL(uint32_t control)
..\Lib\CMSIS\inc\core_cmFunc.h(76): error:  #247: function "__get_IPSR"  has already been defined
  __STATIC_INLINE uint32_t __get_IPSR(void)
..\Lib\CMSIS\inc\core_cmFunc.h(89): error:  #247: function "__get_APSR"  has already been defined
  __STATIC_INLINE uint32_t __get_APSR(void)
..\Lib\CMSIS\inc\core_cmFunc.h(102): error:  #247: function "__get_xPSR"  has already been defined
  __STATIC_INLINE uint32_t __get_xPSR(void)
```

请问这种情况应该如何解决？

dji-dev, 2025-05-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/141641'>Zendesk ticket #141641</a>:<br><br>您好，关于STM32平台的运行，我们在此处有提供对应的指引文档，您可以先了解下：https://developer.dji.com/cn/document/dc0b69f6-e234-4390-87a5-97df7d96612d#%E8%BF%90%E8%A1%8Cstm32-%E7%A4%BA%E4%BE%8B%E4%BB%A3%E7%A0%81
关于您的错误，核心是CMSIS头文件重复定义导致的符号冲突。建议：
检查和确认下项目中是否只包含一个 CMSIS 版本。清理旧文件，避免重复编译。检查包含路径和宏定义，确保一致性。
<br><p><em><small>°°°</small></em></p>